benlye/flash-multi #1: Maple com port error

fremebondo, 2019-07-05:
Just tried to flash a brand new Irangex4 plus.

Maple serial port is found correctly and installed in Windows 7.

When I try to flash i get this output:

Maple device found in USB mode
Starting Multimodule update
Switching Multimodule into DFU mode ... failed!

and verbose output:

.\bin\maple-reset.exe COM32

maple-reset 0.1
ERROR: Failed to open COM32

Any suggestion?

fremebondo, 2019-07-05:
I also tried by launching maple-reset from cmd:

C:\Program Files (x86)\FlashMulti\bin>maple-reset.exe COM32
maple-reset 0.1
This program is Free Sofware and has NO WARRANTY

https://github.com/benlye/maple-reset

ERROR: Failed to open COM32

If i try to open serial port with a terminal (RealTerm) it is opened with no error, so I thing there is some issue with maple-reset.exe

Launching as admin does not solve.

benlye, 2019-07-05:
I'll have to spin up a Windows 7 VM and do some testing.

benlye, 2019-07-05:
Is your system Windows 7 64-bit?

benlye, 2019-07-05:
Found the bug - nothing to do with Windows 7.
https://support.microsoft.com/en-gb/help/115831/howto-specify-serial-ports-larger-than-com9

Updated version coming out later today.

Thanks for reporting it!

benlye, 2019-07-05:
I'm pretty sure I've fixed it but just to be sure can you try replacing your maple-reset.exe with the new one here?
https://github.com/benlye/maple-reset/releases/download/0.2/maple-reset-0.2.zip

Once you confirm that it works I'll publish a new version of Flash Multi.

benlye, 2019-07-07:
Fixed in v0.1.2 or later by updating maple-reset.exe to support COM ports larger than COM9.

closed 2019-07-07 by benlye