benlye/flash-multi #11: URUAV TMX5 Lite - cannot flash.

Tygrys-1, 2019-09-24:
Windows 10, URUAV TMX5 Lite module, usb connection recognized as COM6.

The log looks like this:

Starting Multimodule update via native USB
Switching Multimodule into DFU mode ... done
Waiting for DFU device ... got it
Writing firmware to Multimodule ... failed!
Attempting DFU Recovery Mode.
Waiting for DFU device to disappear ...Waiting for DFU device to appear ... got it.
Writing firmware to Multimodule ... failed!

Verbose output:

.\tools\maple-reset.exe COM6

maple-reset 0.3
This program is Free Sofware and has NO WARRANTY

https://github.com/benlye/maple-reset

Reset sequence sent to COM6.

.\tools\dfu-util.exe -R -a 2 -d 1EAF:0003 -D "D:\Dron\OpenTx\flash-multi-0.2.0\multi-stm-opentx-inv-v1.2.1.81.bin"

Lost Device after reset, assuming prod_id was incremented by oneNo DFU capable USB device found
dfu-util - (C) 2007-2008 by OpenMoko Inc.

This program is Free Software and has ABSOLUTELY NO WARRANTY

.\tools\dfu-util.exe -R -a 2 -d 1EAF:0003 -D "D:\Dron\OpenTx\flash-multi-0.2.0\multi-stm-opentx-inv-v1.2.1.81.bin"

Lost Device after reset, assuming prod_id was incremented by oneNo DFU capable USB device found
dfu-util - (C) 2007-2008 by OpenMoko Inc.

This program is Free Software and has ABSOLUTELY NO WARRANTY

Tygrys-1, 2019-09-24:
Sorted out by installing maple drivers, update now ok.

closed 2019-09-24 by Tygrys-1

benlye, 2019-09-24:
I thought that might be the answer. Glad you got it working.

Tygrys-1, 2019-09-24:
Thank you for the good program. :)