benlye/flash-multi #16: Google Safe Browsing Blocking releases

harking, 2020-03-31:
Tried to access the releases page and ran into this:
https://transparencyreport.google.com/safe-browsing/search?url=https:%2F%2Fgithub.com%2Fbenlye%2Fflash-multi%2Freleases&hl=en-US

benlye, 2020-03-31:
Ugh, that's great.  Fixing this might be very difficult because it requires proving ownership of the site, and there's no way to do that with Github.

I suspect it's probably triggered by the driver installers which are in the downloads, but they are completely safe and only run when the user selects them.

VirtualEnder, 2020-04-11:
Windows is actually registering this file as a trojan. 
[image]

benlye, 2020-04-11:
Sigh.  This has happened before.  Fortunately Microsoft are actually decent and provide a way that normal human beings can get them to re-assess the package.  I've submitted it to them to take a look at and hopefully it will be unblocked in the next Windows Defender signature update.  Normally they turn it around quickly.

I wish I could work out what the cause of all this is.  The installer and app really don't do anything that could be construed as malicious.  The installer does install drivers, but so do many other software packages.

benlye, 2020-04-11:
After a little bit more digging, it seems like the most likely culprit is wdi-simple.exe, which is the installer for the Maple USB drivers.  It's another open source package on Github: https://github.com/pbatard/libwdi/blob/master/examples/wdi-simple.c

Scanning the compiled wdi-simple.exe file on VirusTotal gives all the same erroneous warnings as the Flash Multi package.

I guess I'll start looking for an alternative way to install the Maple drivers.

benlye, 2020-04-22:
I'm going to try removing the Windows installer downloads from the releases to see if that helps with the Google Safe Browsing issue.  I don't know if it will, or if Google will even re-scan the site in order to detect the change.

In the meantime I have mirrored the repository to an new one, and will not be attaching the executable installers in an attempt to see if that is the problem.  

Please go to this repository to download the latest version of Flash Multi:
https://github.com/benlye/flashmulti/releases

benlye, 2020-04-26:
It seems like removing the executable installers has fixed the issue with Google safe browsing :-)

I'll remove the cloned repo.

benlye, 2020-04-26:
Hopefully we've seen the last of this!

closed 2020-04-26 by benlye