benlye/flash-multi #25: Flashing Bootloader fails on macOS

C0br4, 2020-08-23:
Hi, i went through the [described procedure](https://github.com/benlye/flash-multi/blob/master/doc/Linux.md#Mac) and I ended up with the latest firmware on my multimodule. So far so good but when flashing via OpenTX I still get the NoSync message. 

Upon closer inspection I realized that erasing the flash and programming the bootloader failed but programming the firmware went fine.

Using the latest macOS Mojave on a macbook pro retina mid 2012 and libusb 1.0.23 (latest as of writing)

Terminal Output:
```
✘ C0br4@C0br4sMBP  ~/Downloads/flash-multi-0.4-1.3  sudo ./flash-multi -f ~/Downloads/multi-stm-opentx-aetr-inv-v1.3.1.49.bin -p /dev/cu.usbmodem14101 
flash-multi 0.4.3

This program is Free Software and has NO WARRANTY.
https://github.com/benlye/flash-multi/

2020-08-23 21:49:00.482 system_profiler[6608:368630] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2020-08-23 21:49:01.049 system_profiler[6616:368656] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
Multi Firmware Version:    1.3.1.49 (STM32)
Expected Channel Order:    AETR
Multi Telemetry Type:      OpenTX
Invert Telemetry Enabled:  True
Flash From Radio Enabled:  True
Bootloader Enabled:        True
Serial Debug Enabled:      False

Proceed with firmware update? [Y]es or [N]o: Y

Attempting USB upload using dfu-util

[1/2] Resetting to DFU mode...
./tools/macos/maple-reset /dev/cu.usbmodem14101
maple-reset 0.1

This program is Free Sofware and has NO WARRANTY

Reset sequence sent to /dev/cu.usbmodem14101

[2/2] Writing firmware...
./tools/macos/dfu-util -d 1eaf:0003 -a 2 -D "/Users/C0br4/Downloads/multi-stm-opentx-aetr-inv-v1.3.1.49.bin" -R
dfu-util 0.8

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2014 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 1eaf:0003
Run-time device DFU version 0110
Claiming USB DFU Interface...
Setting Alternate Setting #2 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0110
Device returned transfer size 1024
Copying data from PC to DFU device
Download	[=========================] 100%       118812 bytes
Download done.
state(8) = dfuMANIFEST-WAIT-RESET, status(0) = No error condition is present
Done!
Resetting USB to switch back to runtime mode

MULTI-Module firmware flashed successfully.
```

benlye, 2020-08-23:
That script doesn't update the bootloader, it only flashes the firmware, and it looks like it worked perfectly.

If you want to update the bootloader you must run the `multi-bootreloader` script.

I need to update the documentation. The bootloader update script for Mac/Linux is fairly new.

C0br4, 2020-08-24:
Oh man, it's right there in the script name. Everything works as expected.

closed 2020-08-24 by C0br4

benlye, 2020-08-24:
Can't really do it in one script for these modules because it's actually a separate app that needs to run, and you need to unplug/plug the module in before you can write the actual firmware.  Also, the bootloader only needs to be re-written once, so it would slow further updates down unnecessarily.

If it was a module with a built-in FTDI bridge (like a Jumper JP4IN1-SE) the bootloader would get written every time the firmware is, just because that's the way that type of module gets updated.

C0br4, 2020-08-24:
Oh yeah that makes sense. Thanks for taking the time to explain