benlye/flash-multi #28: TMX5 module cannot be flashed because flash size.

geniack, 2020-09-19:
Hello,

I just received a TMX5 module from banggod made by URUAV. I am using 'flash-multi' to try to install a new firmware on it. People on banggood and on the github page of the software developer suggesting to use this file: 

multi-stm-opentx-aetr-inv-v1.3.1.65.bin

So I put my usb mode into 'Sticky DFU mode', updated the bootloader which worked fine. But once I am trying to flash that module I get an error:

Module flash verification failed. This MCU on this module does not support 128KB of flash.

I just opened up the module and it has a STM32 processor which should have at least that amount of flash. 

When I erase the flash, it seems to work and I get this output:

.\tools\dfu-util-multi.exe -a 2 -d 1EAF:0003 -D ".\tools\erase120.bin" -v

dfu-util-multi 0.7

Copyright 2005-2008 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2012 Tormod Volden and Stefan Schmidt
Copyright 2020 Ben Lye
This program is Free Software and has ABSOLUTELY NO WARRANTY

This version of dfu-util has been modified for flashing an STM32 MULTI-Module. It is intended for use with Flash Multi.
Please report bugs to https://github.com/benlye/dfu-util-multi

Filter on vendor = 0x1eaf product = 0x0003
Opening DFU capable USB device... ID 1eaf:0003
Run-time device DFU version 0110
Found DFU: [1eaf:0003] devnum=0, cfg=1, intf=0, alt=2, name="STM32duino bootloader v1.0  Upload to Flash 0x8002000"
Claiming USB DFU Interface...
Setting Alternate Setting #2 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0110
Device returned transfer size 1024
No valid DFU suffix signature
Warning: File has no DFU suffix
bytes_per_hash=2457
Copying data from PC to DFU device
Starting download: [##################################################]
 finished!

Sent a total of 122880 bytes
Done!

To me it looks like 128 KB of flash have been erased. So I am not really sure where to go from here.
Your help is much appreciated.

Kind regards

geniack

benlye, 2020-09-19:
TMX5 modules are well known to have the wrong STM32. Please attach a photo of the STM32 to verify it is an STM32F103CBT6 and not an STM32F103C8T6.

benlye, 2020-09-19:
It's not unusual for a C8 modules to allow you to write 128KB to it, but a chip labelled as a C8 only officially supports 64KB.  

The test we do is to write 128KB then to try reading it - most C8s will only let you read 64KB from the flash, which is what yours must be doing.  A genuine CB module will allow 128KB to be written and read.

We have had to take this step because people with modules with C8 chips have reported unreliable and erratic behavior.

https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/393

geniack, 2020-09-19:
Hello benlye! 

Thank you for the fast reply! I just took a photo of the STM32:

https://imgur.com/a/eeg5mgS

It seems to be indeed a STM32F103C8T6.
I am not able to flash new software?

Kind regards

geniack

geniack, 2020-09-20:
I have read the thread you have linked, seems like I will ask banggood to refund me my money. The question is, can I still use this device I have now? To my understanding the precompiled image I was trying to flash includes all the protocols. If I need for example only flysky, could I get away with compiling a smaller image with only flysky support and flashing this to the module?

Kind regards

geniack

benlye, 2020-09-20:
Our advice is to replace the module.  The reason we've taken these measures is because we've had multiple reports of erratic behavior when using them, which we believe is caused by the MCU/flash.

You could compile your own firmware (it would have to be smaller than 56KB to allow for the 8KB bootloader) but even then the EEPROM resides in the last 2KB of the 128KB flash, so you cannot avoid the module trying to use all of it.

You would also need to disable the flash size check which is now present in the firmware - the details on how to do that are in the configuration file, _config.h.  But again, our advice is to use a module with the correct part.

iz3man, 2020-10-02:
What I

> seems like I will ask banggood to refund me my money

Could you please let me know if you succeded? It would be interesting to share ppl's experience, so we know if it's worth the hassle and wait time. BG support is far from good and/or user friendly.

jogybaer0815, 2020-10-07:
thats is bad to hear - module flashing in my X10S is still possible, but there are missing protocolls No 73 to 77
Is the reason also because of EEPROM flash size or whats the problem?

benlye, 2020-10-07:
> thats is bad to hear - module flashing in my X10S is still possible, but there are missing protocolls No 73 to 77
> Is the reason also because of EEPROM flash size or whats the problem?
> (Fimware .59 is the last functional for me)

You will not be able to run the latest firmware (which has the latest protocols) on your module unless you compile it yourself and disable the flash size check in the firmware.  By default it will refuse to run and the LED will just blink 'SOS'.

benlye, 2020-10-07:
All the options for these unsupported modules are documented here:
https://www.multi-module.org/basics/module-hardware/unsupported-modules

closed 2020-10-07 by benlye