benlye/flash-multi #31: Unable to flash bootloader URUAV TMX5

iz3man, 2020-11-21:
I have several issues with my URUAV when trying to connect to my Frsky D8 whoop. Sometimes it needs a dozen power cycles to connect, sometimes (like today) it's didn't work at all.

I read there might be an issue with the bootloader, so I tried to update it with flash-multi.

The problem I see (and saw as well last time I tried) is, that the RED led NEVER goes out, not even after several minutes.

This is the log:

`.\tools\dfu-util-multi.exe -a 2 -d 1EAF:0003 -D ".\tools\bootreloader_legacy.bin" -v -R

dfu-util-multi 0.7

Copyright 2005-2008 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2012 Tormod Volden and Stefan Schmidt
Copyright 2020 Ben Lye
This program is Free Software and has ABSOLUTELY NO WARRANTY

This version of dfu-util has been modified for flashing an STM32 MULTI-Module. It is intended for use with Flash Multi.
Please report bugs to https://github.com/benlye/dfu-util-multi

Filter on vendor = 0x1eaf product = 0x0003
Opening DFU capable USB device... ID 1eaf:0003
Run-time device DFU version 0110
Found DFU: [1eaf:0003] devnum=0, cfg=1, intf=0, alt=2, name="STM32duino bootloader v1.0  Upload to Flash 0x8002000"
Claiming USB DFU Interface...
Setting Alternate Setting #2 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0110
Device returned transfer size 1024
No valid DFU suffix signature
Warning: File has no DFU suffix
bytes_per_hash=461
Copying data from PC to DFU device
Starting download: [##################################################] finished!

Sent a total of 23084 bytes
Done!
Resetting USB to switch back to runtime mode`

[image]

Before you ask: This is a TMX5 with 128kb STM32 chip. :) I swapped the 64k chip.

benlye, 2020-11-21:
I've seen that happen on a TMX5 module. The process takes less than 30s, so it's OK to unplug the module after 30s even if the LED is still on. 

I'll update the docs.

iz3man, 2020-11-21:
Ok Thanks. I assumed the bootloader to have flashed ok, as I can update the module from inside the transmitter. In case that's what the bootloader is needed for ;)
I heard that a bad bootloader would show those issues I see with D8. And I guess the bootloader being written as of today would solve that problems.
So any other idea what my problem may be? Or better open a case in multi-loader git?

benlye, 2020-11-21:
So you've updated the bootloader and you still have intermittent problems activating D8?  Probably best to ask in the RC Groups forum.
https://www.rcgroups.com/forums/showthread.php?2165676-DIY-Multiprotocol-TX-Module

iz3man, 2020-11-21:
Wanted to avoid that :) 1.3k+ posts monster thread ...

closed 2020-11-21 by benlye