benlye/flash-multi #32: Flash-multi crashes when attempting to "Flash Module Bootloader"

alexbirkett, 2020-11-24:
Steps to reproduce:

* Connect IRangeX via USB
* Open Flash Multi v0.5.1
* Open Advanced -> Actions Menu
* Select "Install USB Device drivers"
* Wait for process to complete
* Click Yes to popups and wait for process to complete:
```
Installing MULTI-Module USB Serial Driver ... done
Switching MULTI-Module into DFU mode ... done
Installing MULTI-Module DFU Bootloader Driver ... done
````
* Unplug IRangeX
* Connect IRangeX
* Open Advanced -> Actions Menu
* Select "Flash Module Bootloader"

Expected results:
* Bootloader is flashed

Actual results
* App crashes (see attached crashlog)

```
See the end of this message for details on invoking 
just-in-time (JIT) debugging instead of this dialog box.

************** Exception Text **************
System.NullReferenceException: Object reference not set to an instance of an object.
   at Flash_Multi.FlashMulti.<UpgradeBootloader>d__70.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at Flash_Multi.FlashMulti.<UpgradeBootloaderToolStripMenuItem_Click>d__50.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()

************** Loaded Assemblies **************
mscorlib
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4180.0 built by: NET48REL1LAST_B
    CodeBase: file:///C:/Windows/Microsoft.NET/Framework64/v4.0.30319/mscorlib.dll
----------------------------------------
flash-multi
    Assembly Version: 0.5.1.0
    Win32 Version: 0.5.1
    CodeBase: file:///C:/Users/oleole/Downloads/flash-multi-0.5.1/flash-multi-0.5.1/flash-multi.exe
----------------------------------------
System.Windows.Forms
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4150.0 built by: NET48REL1LAST_C
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms/v4.0_4.0.0.0__b77a5c561934e089/System.Windows.Forms.dll
----------------------------------------
System
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4001.0 built by: NET48REL1LAST_C
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System/v4.0_4.0.0.0__b77a5c561934e089/System.dll
----------------------------------------
System.Drawing
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.3752.0 built by: NET48REL1
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Drawing/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Drawing.dll
----------------------------------------
System.Configuration
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.3752.0 built by: NET48REL1
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Configuration/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Configuration.dll
----------------------------------------
System.Core
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4180.0 built by: NET48REL1LAST_B
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Core/v4.0_4.0.0.0__b77a5c561934e089/System.Core.dll
----------------------------------------
System.Xml
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.3752.0 built by: NET48REL1
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Xml/v4.0_4.0.0.0__b77a5c561934e089/System.Xml.dll
----------------------------------------
System.Management
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.3752.0 built by: NET48REL1
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Management/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Management.dll
----------------------------------------

************** JIT Debugging **************
To enable just-in-time (JIT) debugging, the .config file for this
application or computer (machine.config) must have the
jitDebugging value set in the system.windows.forms section.
The application must also be compiled with debugging
enabled.

For example:

<configuration>
    <system.windows.forms jitDebugging="true" />
</configuration>

When JIT debugging is enabled, any unhandled exception
will be sent to the JIT debugger registered on the computer
rather than be handled by this dialog box.

```

alexbirkett, 2020-11-24:
Is there any way to install the bootloader on a Mac?

benlye, 2020-11-27:
Strange - never seen any problems like that before.  I'll try to see if I can figure out what's going on there.

Yes, there is a Linux/Mac version of the bootloader update tool distributed in the Mac/Linux download - [multi-bootreloader](https://github.com/benlye/flash-multi/blob/master/linux/multi-bootreloader)

Getting it to work on a Mac (especially a recent one) isn't super easy, but it is possible.
https://github.com/benlye/flash-multi/blob/master/doc/Linux.md

If that looks like too much work I also have a script I can give you to run on Windows.

alexbirkett, 2020-11-28:
I attempted to flash the IRangeX using Taranis x7 and that also failed. Then I kept the IRangeX in the Taranis and also connected a TTL-USB module because ["it is strongly recommended that you power your module from the transmitter when flashing it using a USB-to-serial adapater."](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Compiling_STM32.md#usb-to-serial-adapter) I don't understand why this is necessary if the TTL-USB module is 3.3v. Mine was actually putting out 3.6v so it did it anyway. 

I built the DIY-Multiprotocol-TX-Module source using the Arduino IDE and the flashing process started by then failed part way through. (I tried this multiple times) but this fixed the bootloader so that I when I attempted to flash again with the Taranis x7 it worked. 

I've had most success flashing STM32 microcontroller with an ST-LINK clone [something like this](https://www.banggood.com/no/Geekcreit-3_3V-5V-XTW-ST-LINK-V2-STM8-STM32-Simulator-Programmer-Downloader-Debugger-500MA-Fuse-Short-Circuit-Protection-Aluminum-Alloy-U-Disk-Shell-With-20cm-Dupont-Wire-p-1177014.html) but the pins to do it don't seem to be broken out on the IRangeX. 

Feel free to close this issue

benlye, 2020-11-28:
Strange. Sounds like maybe the USB-to-serial adapter is a bit suspect. Normally the IRX4 modules are very robust and easy to flash, especially via serial.

Glad you got it sorted eventually.

closed 2020-11-28 by benlye