benlye/flash-multi #33: Flash-Multi 05.1 can't use Windows USB ports

DonBrewster, 2021-01-07:
Tried to use Flash-Multi 05.1 app to update IR4X plus (at least a couple of years old) firmware. The USB drivers will not load. Windows says there is a device descriptor failure. Tried this on two different Windows 10 PCs with the same result. I also tried to update the IR4X plus through a new Horus X10s and it failed with a no sync error. I can use the IR4X plus in my Taranis 9d+, but just can't get it updated. Any suggestions?

benlye, 2021-01-07:
What happens when you try to install the device drivers?

DonBrewster, 2021-01-07:
When I try to install the USB drivers, it goes through the process just fine. But I never see a com port or DFU or anything indicating I can communicate with the IRX4 module.

benlye, 2021-01-07:
Maybe you could post some screenshots? Verbose logs of the driver installation, screen shots of device manager?

Have you tried different USB cables? 

The driver installer works fine for most people, including me, so you're going to have to give me more to go on if you want help.

DonBrewster, 2021-01-07:
Here are a few pictures. One is the IRX4. I've tried more than one cable. Another two is the USB error messages. A fourth is the USB drivers loaded from the app. And the last shows the options to download or upload are grayed out buttons. I'll provide anything you think will help. Thank you. 
[image]
[image]
[image]
[image]
[image]

benlye, 2021-01-07:
Do you see progress bars when the drivers are being installed? It should take a couple of minutes and there should be progress bars.

You could try running the driver installer commands manually from a command prompt.

benlye, 2021-01-07:
What are the LEDs on the module sounds when it is plugged into the computer?

benlye, 2021-01-07:
I guess the other option is that the drivers are installed just fine, but the USB port on your module is faulty. You could open it up and check the soldering.

DonBrewster, 2021-01-08:
Thanks for the feedback. I'll get back to this soon and let you know what I find.

DonBrewster, 2021-01-08:
The progress bars move until the drivers complete their installation. Everything looks normal. 
The Green LED is steady on and the Red LED blinks at about a one second on, one second off rate.
I looked over the USB connector on the module, and it looks fine. I suppose the USB circuitry could be bad.
The USB-to-serial install is probably the option I have left. Thanks for the assistance.

DonBrewster, 2021-01-12:
Benlye, It turns out the module had no bootloader on it. I ended up having to use the Arduino IDE and an FTDI to direct load to the STM board. I now have a bootloader and everything appears to be correct. Thanks.

DonBrewster, 2021-01-13:
Benlye, I still have an issue if you can help. As I stated, I used the Arduino IDE and FTDI to load a bootloader. The version was then readable (and I think it was v1.3.2.1, but I didn't write it down). So I used the Flash-Multi 05.1 and loaded the latest (v1.3.1.92) version to the IRX4 via the USB port. Everything appeared to go correctly and I can see the correct version in my Taranis. But now I bind to a Blade Nano QX and I get no throttle response. It did work correctly on the module before the update. The Taranis channel display shows the throttle channel changing. The only channel that appears to be working is the mode changing switch between agility and stability mode (led changes color). The file I used was multi-stm-serial-aetr-v1.3.1.92.bin. At least I have a bootloader now to reload the firmware. Is it possible the bin load was corrupted?

benlye, 2021-01-13:
It's unlikely that the firmware was corrupt. I don't know anything about that quad, but as it uses DSMX it's most likely a channel order problem, or a problem with the throttle channel not being low enough for the quad to arm.

You should ask on the main MULTI-Module Git repository, or in the RC Groups forum.

closed 2021-01-14 by benlye