benlye/flash-multi #34: USB Supporting Warning Message

dayue1984, 2021-01-09:
I've updated my irangx 4in1 to the new bootloader, then I want to update the firmware, and always get this message, but where is the OK button? I can‘t even close this MsgBox. Thanks!
[image]

benlye, 2021-01-09:
What version of Windows are you running? What language is your OS?

dayue1984, 2021-01-09:
> What version of Windows are you running? What language is your OS?

I'm running the latest Windows10 20H2, system display language is Simplified Chinese. Thanks!

benlye, 2021-01-09:
Thanks, I'll try to reproduce.

benlye, 2021-01-09:
I think I've found the problem.  I need to work on a better solution and maybe use a different font which has a better fallback on non-English systems, but this test version should display the dialog box correctly for you so you can update your module.

[No-Font-Scaling.zip](https://github.com/benlye/flash-multi/files/5791736/No-Font-Scaling.zip)

(The only other change from 0.5.1 is that the bootloader type option is removed - the new 'Sticky DFU' bootloader is always enabled.)

dayue1984, 2021-01-09:
It’s working, thanks alot!

benlye, 2021-01-17:
Fixed in latest release.

closed 2021-01-17 by benlye