benlye/flash-multi #42: the new version shows as infected by bitdefender

tj51, 2021-03-21:
the program shows as malware infection it really sucks since it a good little program

benlye, 2021-03-21:
It's a false positive, there's nothing I can do.  You can go through the source code if you want to - there's nothing malicious there.

benlye, 2021-03-21:
If you want to you can submit a false positive report.  The report needs details of the actual detection, so I cannot do it as I don't have that product installed.
https://www.bitdefender.com/consumer/support/answer/40673/

closed 2021-03-21 by benlye

tj51, 2021-03-21:
I scanned the zip file, then the extracted folder, nothing, opened the 
program loaded bin from the multi get hub page, clicked on write, then 
the fun started,

I would send to bit defender if you program stop trying to communication 
somewhere when just sitting idle,

when bitdefender deleted the utility the dfu starts trying to 
communicating, anyone would call this a virus,

can you fix the program from doing that behavior, thank you, Tj

benlye, 2021-03-21:
I'm not going to do anything. There is no malware in the program or any of the utilities it uses.