benlye/flash-multi #43: Issue with FrSKY Xlite Pro and IRangeX Multiprotocol Module

crstecker, 2021-03-29:
Hi,

I upgraded my FrSKY Xlite Pro and now my multiprotocol module is no longer accessible from the radio. The radio no longer sees it. These are the firmware version I am using:

FrSKY Xlite Pro is now running 2.3.12
The ISRM-Pro is now on 1.1.0 / 2.1.6 FCC
The multiprotocol module (IrangeX IRX4 Lite) is showing Serial Port DFU Device in Flash Multi  v0.6.1. The Firmware shows 1.3.2.61 in Flash Multi. I also tried 1.3.2.58, 1.3.2.30, 1.3.2.12 and that didn't work either.

I am not sure what to try next. The multiprotocol module worked great before it was flashed.

Thank you for the help.

benlye, 2021-03-29:
I don't know anything about that radio. Have you updated the module bootloader?

Chris15X, 2021-03-29:
OK thank you. Sorry I meant to post on the Multi-Module page. Yeah I don't think it is Flash Multi either. I did update the bootloader. 

Thank you for the quick response.

Have a great day.

closed 2021-03-29 by benlye