benlye/flash-multi #44: You MUST update the MULTI-module bootloader to the latest version BEFORE writing this firmware

HNeumeier, 2021-05-12:
can you help me i am working on Linux and want to flash the newest firmware to an IRangeX Modul
but i get the following errors and i don't now how to flash the bootloader.

helmut64@helmut64-Lubuntu:/home/daten/text/Modellbau/Taranis/Multiprotokoll_Module/Firmware/flash-multi-0.6.0$ sudo ./flash-multi -l -f multi-stm-serial-aetr-v1.3.2.77.bin  -p /dev/ttyACM0 
flash-multi 0.6.0

This program is Free Software and has NO WARRANTY.
https://github.com/benlye/flash-multi/

Firmware File Name:        multi-stm-serial-aetr-v1.3.2.77.bin
Multi Firmware Version:    1.3.2.77 (STM32F1)
Firmware Target:           STM32F103CB (128KB)
Expected Channel Order:    AETR
Multi Telemetry Type:      OpenTX or erSkyTx (MULTI_TELEMETRY)
Invert Telemetry Enabled:  True
Flash From Radio Enabled:  True
Bootloader Enabled:        True
USB Serial Support:        False
Serial Debug Enabled:      False

Proceed with firmware update? [Y]es or [N]o: y

Attempting USB upload using dfu-util

WARNING: Specified firmware file was not compiled with USB support.
         You MUST update the MULTI-module bootloader to the latest version BEFORE writing this firmware.

         See https://github.com/benlye/flash-multi/blob/master/doc/New_Bootloader.md for more information.

Proceed with firmware update? [Y]es or [N]o: n

Firmware update aborted.

benlye, 2021-05-12:
Use the `multi-bootreloader` script. It should be located in the same folder as `flash-multi`.

HNeumeier, 2021-05-12:
Thanks, now it works.

closed 2021-05-12 by HNeumeier