benlye/flash-multi #48: Flashing New Bootloader on Mac

riverbasin, 2022-01-16:
Hi There,

I have a new IRangeX from Banggood.  I am attempting to flash it with the latest firmware on a Taranis X9D+ 2019 from a Mac M1 running Monterey.  It looks like the IRangeX does not have the latest bootloader, so the flash fails with the following warning:

Attempting USB upload using dfu-util

WARNING: Specified firmware file was not compiled with USB support.
         You MUST update the MULTI-module bootloader to the latest version BEFORE writing this firmware.

         See https://github.com/benlye/flash-multi/blob/master/doc/New_Bootloader.md for more information.

Unfortunately, I do not see instructions for flashing the new bootloader on a Mac (or Linux) on the New_Bootloader page.  Can you please advise on the procedure for updating the bootloader from a Mac?

Thanks!

benlye, 2022-01-16:
You're right, it's not documented, but there is a tool - if you download the [Linux scripts](https://github.com/benlye/flash-multi/releases/download/0.6.1/flash-multi-0.6.0.tar.gz) there is a script named `multi-bootreloader` which can be run on Linux/macOS and will install the new bootloaer.

riverbasin, 2022-01-16:
Thanks @benlye.  That seemed super promising when it started to run, but now seems hung. The LED is still continuously blinking.  Output listed below - I'm hesitant to Ctrl-C or interrupt due the the dire warning prior to "Yes."

``Attempting USB upload using dfu-util

[1/2] Resetting to DFU mode...
./tools/macos/maple-reset /dev/tty.usbmodem101
maple-reset 0.1

This program is Free Sofware and has NO WARRANTY``

benlye, 2022-01-16:
At that point you can safely Ctrl-C and kill it - nothing has been written to the module yet.

Once you've killed it you can run the `maple-reset` command manually to try to see what's going on.  I think if you can get the module into DFU mode with maple-reset you can just re-run `multi-bootreloader` again and it will skip that step and go straight to installing the bootloader app.

closed 2022-02-05 by github-actions[bot]