benlye/flash-multi #49: not working for me despite trying

majorweaknesss, 2022-01-26:
ive got the taranis x7.
all firmware updated. followed the instructions to the letter and no joy.
at one point updated usb drivers for maple via zadig.
com port seems to drop out at the point of read/write.

have updated usb drivers successfully, have been able to boot into DFU mode, and have been able to read the flash. 

but when it all seems good, i attempt to write and get a 'non usb compiled flash- update usb drivers' warning.
so rinse and repeat.

my other option is updating via SD card, but i dont have an SD card on me right now.

But curious as to why this isnt working for me.

benlye, 2022-01-26:
Can you post a screenshot of the error you're getting?

benlye, 2022-01-26:
Also, can you say which external module you have, and describe exactly what the module LEDs do when you plug it in via USB?  This will tell me what bootloader is installed.  I expect that the message you refer to above is actually warning you that you may need to update the bootloader.

closed 2022-02-15 by github-actions[bot]