benlye/flash-multi #50: Failed to read the multi module!

DerP4si, 2022-04-13:
If i try to do anything with multiflash (reading, writing) it always ends with 
"Reading Flash Memory...failed!" "Failed to read the MULTI module!"
is it broken or am i doing something wrong?
Thanks <3
[image]

benlye, 2022-04-13:
Nine times out of ten this is a driver problem.  What module do you have?

DerP4si, 2022-04-13:
iRangeX IRX4 Plus 2.4G

benlye, 2022-04-13:
Can you enable Verbose Output and show the verbose log?

benlye, 2022-04-13:
Also, make sure that the module is plugged directly into your computer, not via a USB hub.  Older USB hubs can cause problems like this.

benlye, 2022-04-13:
Never mind - you're using the right driver (libusbK) anyway.  Next thing is to make sure that the module is connected directly to the computer and not via a USB hub.  After that, try using different USB cables.

DerP4si, 2022-04-13:
Well i am plugged directly in the mb and tried another cable... same issue. Gonna try it on my notebook next...

DerP4si, 2022-04-13:
same problem

DerP4si, 2022-04-14:
Ok i think its the Module itself... even with windows 7 its the same problem...tho I am curious why it says Multi Module Bootloader in the device manager

benlye, 2022-04-14:
The module is giving the correct USB device ID, which is how Device Manager identifies it.  The problem seems to be coming from actually trying to communicate with it - it's not sending back the right information to the programming utility.  

It's pretty rare for one of these things to have a faulty USB interface, but it does happen.

DerP4si, 2022-04-14:
Ok ^^ my goal at the end is just to get it in ppm standalone mode to use it like this:
https://youtu.be/rYxwIrqE7q8
(on min 56 he shows that he installs the multi-stm-ppm-aetr-v1.3.2.86.bin firmware to do so)
it seems the shop i bought it installed the firmware 1.3.3.0... is there any way to get into ppm mode anyway or have i to refund it? ^^ Thanks for your quick replys btw

benlye, 2022-04-14:
If you can get the shop to replace it, that might be a good idea as it would probably be easiest.

If not, there are still ways you can program it:
1. From a radio
  Any radio which runs OpenTX/EdgeTX or erSky9x/erSkyTX and supports the MULTI-Module can update the firmware on a the module.  If you have one or know someone who does then you can go this route pretty easily.

2. Using an external USB to Serial adapter
  You would need to solder some header pins onto the module's PCB and attach at to your computer via a USB-to-serial adapter such as a CP2102 like [this](https://www.amazon.co.uk/UART-CP2102-Module-Serial-Converter/dp/B00AFRXKFU).  This method is a bit more hassle but completely bypasses the on-board USB interface.  It's fiddly to get set up, but as long as you get the soldering and connections right it will work.

DerP4si, 2022-04-14:
Yeah i thought so... He just flashed the right firmware and gonna send me the module right away...
Thanks for your help anyway i hope it works now as i intend so that i can finally have a RC Car again ^^

closed 2022-05-03 by github-actions[bot]