benlye/flash-multi #51: All operations failing

sansjunk, 2022-09-24:
I just got a new iRangeX IRX4 Plus module from ebay. EdgeTX shows that is running "V1.2.0.22". 

When plugged into the PC via the USB cable the device shows up as "MULTI-Module USB Serial (COM6)". Looks like the device never enters DFU mode. The red LED flashes about once a second and green LED is off. I tried reinstalling the drivers, both using the Flash Multi app as well as Zadig. Reseting the module to DFU doesn't do anything. None of the operations work. Here is the output of "Read Module" command along with the verbose log.

```
Reading MULTI-Module via native USB
Switching MULTI-Module into DFU mode ... done
Reading flash memory ... failed!
```

Verbose log:
```
.\tools\maple-reset.exe COM6 2000

maple-reset 0.5
This program is Free Sofware and has NO WARRANTY

https://github.com/benlye/maple-reset

Reset sequence sent to COM6
Waiting for DFU device ........... got it.

Device reset successful in 393ms.

.\tools\dfu-util-multi.exe -a 2 -d 1EAF:0003 -U "C:\Users\Santosh\AppData\Local\Temp\tmpA965.tmp" -v

dfu-util-multi 0.7

Copyright 2005-2008 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2012 Tormod Volden and Stefan Schmidt
Copyright 2020 Ben Lye
This program is Free Software and has ABSOLUTELY NO WARRANTY

This version of dfu-util has been modified for flashing an STM32 MULTI-Module. It is intended for use with Flash Multi.
Please report bugs to https://github.com/benlye/dfu-util-multi

Filter on vendor = 0x1eaf product = 0x0003
```

Looks like the module is missing proper USB support in the bootloader? Would I need to use a USB to serial adapter to get it going?

benlye, 2022-09-26:
It looks like the module is getting switched to DFU mode correctly.  What is appearing in Device Manager when the device is in DFU mode?  Can you post a screenshot showing the DFU device?

sansjunk, 2022-09-26:
The device manager flashes for a second as if the device is unplugged and plugged back in. Then it shows the same device "MULTI-Module USB Serial (COM7)" as when initially plugged in. You can disregard the com port number since this run the com port changed. Looks like it's not staying in DFU mode?

[image]

benlye, 2022-09-26:
Hmm, that's a new one - I've not seen a device do that before.  Does the LED pattern of the module change at all?  Slow blinking of the red LED generally means that it's in 'normal' mode, awaiting a signal from the radio.  Does it change to rapid blinking, even briefly, after the maple reset command?

I'm thinking you'll probably have to use a USB-to-Serial adapter to re-flash the bootloader to get it working properly.

sansjunk, 2022-09-26:
When I send the reset command the red LED flashes fast for about a second, then flashes medium for the next and then back to it's slow rate of once per second. Yes, I am going to try the solder the header and try flashing using the USB-to-serial adapter. The adapter I have on had doesn't do 3.3v voltage level so will have to wait for the correct one.

benlye, 2022-09-26:
OK, so it goes to DFU mode for at least a short time.  You can try to run the commands manually in a command prompt if you want:

`.\tools\maple-reset.exe COM6 2000`

Then as soon as you see the medium flashing send the dfu-util command:
`.\tools\dfu-util-multi.exe -a 2 -d 1EAF:0003 -U "C:\Users\Santosh\AppData\Local\Temp\tmpA965.tmp" -v`

You may have to experiment with the timing a bit to get it to work, but it should be possible to execute the second command.  Once the DFU command is running the device should stay in DFU mode until it is complete / indefinitely.

benlye, 2022-09-26:
OK, looks like flashing via the serial pins is going to be the way to go. When you flash that way the bootloader will be reinstalled automatically any time you write new firmware, so you just need to write the new firmware and you should be done.

sansjunk, 2022-09-26:
You just read my mind, I was wondering exactly that - if I should flash the bootloader first. I will report back here shortly on how it went.

sansjunk, 2022-09-27:
I used a level converter with the USB-to-serial adaptor and flashed the latest firmware to the IRange module. Everything works expected now. Thanks!

closed 2022-09-27 by sansjunk