benlye/flash-multi #52: If you can get the shop to replace it, that might be a good idea as it would probably be easiest.

adejo88, 2022-09-29:
If you can get the shop to replace it, that might be a good idea as it would probably be easiest.

If not, there are still ways you can program it:
1. From a radio
  Any radio which runs OpenTX/EdgeTX or erSky9x/erSkyTX and supports the MULTI-Module can update the firmware on a the module.  If you have one or know someone who does then you can go this route pretty easily.

2. Using an external USB to Serial adapter
  You would need to solder some header pins onto the module's PCB and attach at to your computer via a USB-to-serial adapter such as a CP2102 like [this](https://www.amazon.co.uk/UART-CP2102-Module-Serial-Converter/dp/B00AFRXKFU).  This method is a bit more hassle but completely bypasses the on-board USB interface.  It's fiddly to get set up, but as long as you get the soldering and connections right it will work.

_Originally posted by @benlye in https://github.com/benlye/flash-multi/issues/50#issuecomment-1099116487_

adejo88, 2022-09-29:
Please just bought a TX16S radio but the internal module is showing NO MULTI_TELEMETRY so i tried to flash but it fails

benlye, 2022-10-04:
> Please just bought a TX16S radio but the internal module is showing NO MULTI_TELEMETRY so i tried to flash but it fails

Exactly which model of the TX16S have you bought?

benlye, 2022-10-04:
This project is only really about updating *external* MULTI-Modules.  Your radio has an internal module so you cannot really use this software.  You should find a TX16 support group, or contact their support directly.

closed 2022-10-04 by benlye