benlye/flash-multi #53: Unable to flash bootloader on iRangeX IRX4+

kaeuferrot, 2023-07-22:
When trying flashing a new bootloader with Flash Multi 0.6.1 I get errors as follows.

Upgrading MULTI-Module Bootloader via native USB
Switching MULTI-Module into DFU mode ... done
Writing BootReloader to MULTI-Module ... failed!
Attempting DFU Recovery Mode.
Waiting for DFU device to disappear ... gone.
Waiting for DFU device to appear ... got it.
Writing BootReloader to MULTI-Module ... failed!!

.\tools\maple-reset.exe COM6 2000

maple-reset 0.5

This program is Free Sofware and has NO WARRANTY

https://github.com/benlye/maple-reset

Reset sequence sent to COM6
Waiting for DFU device .......... got it.

Device reset successful in 392ms.

.\tools\dfu-util-multi.exe -a 2 -d 1EAF:0003 -D ".\tools\bootreloader_stickydfu.bin" -v -R

dfu-util-multi 0.7

Copyright 2005-2008 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2012 Tormod Volden and Stefan Schmidt
Copyright 2020 Ben Lye
This program is Free Software and has ABSOLUTELY NO WARRANTY

This version of dfu-util has been modified for flashing an STM32 MULTI-Module. It is intended for use with Flash Multi.
Please report bugs to https://github.com/benlye/dfu-util-multi

Filter on vendor = 0x1eaf product = 0x0003
Opening DFU capable USB device... ID 1eaf:0003
Run-time device DFU version 0110
Found DFU: [1eaf:0003] devnum=0, cfg=1, intf=0, alt=2, name="UNDEFINED"
Claiming USB DFU Interface...
Cannot claim interface
.\tools\dfu-util-multi.exe -a 2 -d 1EAF:0003 -D ".\tools\bootreloader_stickydfu.bin" -v -R

dfu-util-multi 0.7

Copyright 2005-2008 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2012 Tormod Volden and Stefan Schmidt
Copyright 2020 Ben Lye
This program is Free Software and has ABSOLUTELY NO WARRANTY

This version of dfu-util has been modified for flashing an STM32 MULTI-Module. It is intended for use with Flash Multi.
Please report bugs to https://github.com/benlye/dfu-util-multi

Filter on vendor = 0x1eaf product = 0x0003
Opening DFU capable USB device... ID 1eaf:0003
Run-time device DFU version 0110
Found DFU: [1eaf:0003] devnum=0, cfg=1, intf=0, alt=2, name="UNDEFINED"
Claiming USB DFU Interface...
Cannot claim interface

thank you
Herbert

benlye, 2023-07-22:
I'm pretty sure you have a drive problem.  See the solutions for updating your Maple drivers here:
https://github.com/benlye/flash-multi/blob/master/doc/Troubleshooting.md

kaeuferrot, 2023-07-22:
Thank you, it helped!

kaeuferrot, 2023-07-22:
Solved.

closed 2023-07-22 by kaeuferrot