benlye/flash-multi #58: windows 11 driver question

beau76, 2024-07-10:
Hi I had the radio master multi  4 in1  external I am running Windows 11 and when I look in device manager you can see how the device is showing up what do I need to install to make this work I am trying to update the boot loader on my multi-module I have tried installing the drivers this is the software that they did not seem to do anything

[image]

benlye, 2024-07-10:
You need to install the Silabs CP210x driver.

https://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers

beau76, 2024-07-10:
Do I want the vcp driver or do I want the universal

benlye, 2024-07-10:
Universal

beau76, 2024-07-10:
Does this mean the module already has the new bootloader on it or do I need to flash it via the USB for the bootloader to be changed?

when I do the boot loader When you say unplug do you mean just the USB cable  or does the whole module have to be unplug from the radio

[image]

benlye, 2024-07-11:
On a module like yours you don't need to independently update the bootloader - it gets rewritten every time you flash the firmware.

All you need to do is flash the latest (or any) firmware and the bootloader is also written.

It probably already has the latest bootloader on it, but updating it one time with Flash Multi week make sure.

closed 2024-07-11 by beau76