benlye/flash-multi #59: Irangex nano pl18 edgetx

swemiklan, 2024-08-21:
Flasht edgetx to pl18 ( no hardware mod ) installd new nano module and tryed to bind. no luck.
Edgetx show ( no multi telemetry detected ) in model page and radio information.
Flash module wiht 1.3.3.33 managede after some trubble dfu mode/ usb i can read firmware of module trou flashmulti 6,1
still no luck binding, sam text in edgetx. Now i am not able to read module wiht flashmulti, module led are solid green and red slow blink. Elrs module works fine.
Can i erase module in flashmulti when its not responding/ able to read module. Other suggestions?
Bangood Note:
"The STM32 MCU have change to ARM, fireware is ARM CORTEX-M
The new stock will be ARM, Existing inventory is not affected"
Mikael

benlye, 2024-08-21:
Assuming you are on Windows, what device shows up in device manager when you plug the module in to your computer? You may need to update the bootloader.

You can try the troubleshooting page for some ideas:
https://github.com/benlye/flash-multi/blob/master/doc/Troubleshooting.md

swemiklan, 2024-08-22:
I did update the bootloader and managed to update to 1.3.3.33 and could read 1 time in flashmulti then not. i was trying in transmitter and no change ( no multi telemetry detected ) message, back to flashmulti and module not recognized by computer.

benlye, 2024-08-22:
You can't do anything if the computer isn't recognising the module. What is showing up in device manager when you plug it in? Can you post a screenshot?

swemiklan, 2024-08-23:
[image]

benlye, 2024-08-23:
>.\tools\dfu-util-multi.exe does not exist

You don't have all the files unzipped correctly. Download the .zip file again and extract all the files, preserving the file/folder structure.

swemiklan, 2024-08-23:
I am running window 10/11 is there a usb issue?  
Serial port maby not conecting propperly in tx, hardware issue?
Any tips on how to troubleshoot hardware ( serial port )?

swemiklan, 2024-08-23:
Witch file to extract to where?  I am a bit lost now?

benlye, 2024-08-23:
You have clearly downloaded FlashMulti from this site at some point, but you have not correctly extracted/expanded the .zip file (it contains the flahs-multi.exe file and also many other dependency files).

Download this file:
https://github.com/benlye/flash-multi/releases/download/0.6.1/flash-multi-0.6.1.zip

Right-click the downloaded file and choose "extract all"

Run flash-multi.exe from with the extracted folder.

closed 2024-08-27 by benlye