benlye/flash-multi #61: Failed to read the multi module!

Cpl-Agarne, 2025-05-22:
If i try to do anything with multiflash (reading, writing) it always ends with
"Reading Flash Memory...failed!" "Failed to read the MULTI module!"
The module is working ok but the firmware is over 4 yrs old so I'm trying to update the FW. What do you think is going on?

Then main window output:
Reading MULTI-Module via native USB
Reading flash memory ... failed!

The verbose output is:
dfu-util-multi 0.7

Copyright 2005-2008 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2012 Tormod Volden and Stefan Schmidt
Copyright 2020 Ben Lye
This program is Free Software and has ABSOLUTELY NO WARRANTY

This version of dfu-util has been modified for flashing an STM32 MULTI-Module. It is intended for use with Flash Multi.
Please report bugs to https://github.com/benlye/dfu-util-multi

Filter on vendor = 0x1eaf product = 0x0003
Opening DFU capable USB device... Cannot open device

Thanks

benlye, 2025-05-24:
Almost certainly still a driver problem.

Try both the solutions here:
https://github.com/benlye/flash-multi/blob/master/doc/Troubleshooting.md#re-installing-the-maple-dfu-device-drivers

Cpl-Agarne, 2025-05-26:
Attempting to reset maple DFU Zadig fails with "unable to open the port com3".
The DM shows the MM DFU driver version as 3.0.7.0 and the location as: Port_#0003.Hub_#0001.

Full message:
C:\Users\CplA\Documents\RC\Multi Module\flash-multi-0.6.1\flash-multi-0.6.1>.\tools\maple-reset.exe COM3
maple-reset 0.5
This program is Free Sofware and has NO WARRANTY

https://github.com/benlye/maple-reset

ERROR: Failed to open COM3

Same result for Com1 and Com0003

Also I'm showing 3 different versions of the firmware for my device with no docs on what the differences are.
iRangeX iRX4 lite (STM32 chip):
mm-stm-serial-aetr-air-v1.3.4.31.bin
mm-stm-serial-aetr-lbt-v1.3.4.31.bin
mm-stm-serial-aetr-sfc-v1.3.4.31.bin

I've opened them all up in the DFU app and their sheets are all exactly the same except for the 3 letters just before the version number in the name. Do you have any insight on which one is the right one?

Cpl-Agarne, 2025-05-26:
> What shows up in Device Manager when the module is plugged in? See attached:

[image]

> The differences between the firmware types are explained on the download site - https://downloads.multi-module.org
Ok I didn't see it in the docs explanation of what the different parts of the file name represented but I was able to figure it out from the drop down boxes that that 3 letter code is for air, surface and EU. I'm good to go on that now.

Cpl-Agarne, 2025-05-26:
Probably some windows 11 weirdness.

Is com3 the correct port in that instance?

Is ibusbk (v3.0.7.0) the version I need or should I update to winusb (v6.1.7600.16385)?

If I can't figure this out it's not the end of the world. For now it's still working (with somewhere north of 4 yr old FW). I'm down to only 3 models that are not Frsky Rx that I use the MM with and when it stops working that will be the nudge I need to go all Ethos. But for now I'm indulging my OCD.

Thanks